rdpoor/cp #45: SensorControl fails first time after instrument plugged in

rdpoor, 2018-02-28:
On Windows, after you plug in an instrument and launch SensorControl, it will fail with "instrument not found" and then close the windows.  Subsequent launches of SensorControl work properly.

rdpoor, 2018-02-28:
The cause: on Mac, the USB interface starts out powered up.  On Windows, the USB interface starts powered down, and doesn't power up until SensorControl expressly calls instrument().power_up().

The fix: add an explicit call to power_up() as follows:
```
    def process_loop(self):
        self.instrument().power_on()
        while True:
            self.instrument().assure_interactive_mode()
            self.process()
```

rdpoor, 2018-03-04:
The fix was somewhat more detailed than what's shown above.  Re-opening.

rdpoor, 2018-03-04:
Traced the problem to the low-level Windows read() function -- for reasons unknown, it wasn't returning data from the instrument.  And if I re-started the program without quitting pdb, the problem persisted: I had to quit out of pdb and restart before the read() function would return data.

On a hunch, I tried closing and re-opening the serial port (if the first ping() attempt failed).  This worked, so it's a reasonable workaround.

Marking fixed.

closed 2018-03-04 by rdpoor

SensorexNPD, 2018-03-07:
Still seeing this on my machine.

rdpoor, 2018-03-07:
Peculiar.  I tested it on Windows 10 / 64 bit.  Is that your environment
too?  And just to make sure: When you boot SensorControl, does it announce
v1.6b as follows:

    SensorControl v1.6b: Copyright © 2018 Sensorex Corporation, all rights
reserved.

?