rdpoor/cp #48: Should be able to set time constant of filters

rdpoor, 2018-03-04:
For diagnostics and calibration, a service tech should be able to adjust the time constant of the smoothing filters.  Current fixed at 30 seconds -- that may be excessive.

rdpoor, 2018-03-06:
Fixed in

closed 2018-03-06 by rdpoor

SensorexNPD, 2018-03-07:
Rob,
Just fired up SensorControl v1.6a.  I do not see how to get to the filter settings.

rdpoor, 2018-03-07:
Well!  That's probably because I haven't created any user-facing method for setting the filter setting.  :)

My main goal was to make sure that the filter settings could be set with a minimum of changes.  I'll add a user-facing method in the next release.

In the meantime, you can set the filter time constant as follows:

1. Create a file -- let's assume it's called filter_constant.json
2. Copy and past the following into the file:

    {
      "IntegrationConstant": 0.003525980657
    }

3. Load the filter_constant.json file into the instrument using the `Load Configuration From File` command (command 13).  After the instrument resets, it will use that value for the filter coefficients.

Some values that might be useful.  The # of seconds given refers to how long it takes for the filters to settle within 0.5% of their final value:

30 seconds: 0.00352598
15 seconds: 0.007039529
7.5 seconds: 0.0140295
3.75 seconds: 0.027862178

The generalized formula is:

    tolerance = 0.995 (half percent)
    update_rate = 50.0 (hz)
    coefficient = 1.0 - pow(1.0 - tolerance, 1.0/(update_rate * settling_time))

And yes, I'll create a new command for SensorControl to set that rather than have to poke it in via the mysterious `Load Configuration from File` approach described above.

SensorexNPD, 2018-03-07:
OK, I will work with what you have here. When github said the issue was closed, I though you had added the code for filter selection.

No need to do that now.

Thanks,

Tony Pipitone
VP of Technology
Sensorex

  1.  Load the filter_constant.json file into the instrument using the Load Configuration From File command (command 13). After the instrument resets, it will use that value for the filter coefficients.